Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #3: T5 v2.3 2.13 Board Turn off LED

misaalanshori, 2020-11-11:
I have this board and would like to have it very low power. Is it possible to disable the Blue LED or will I need to just desolder it? It would be great if there is another way to disable it as I don't trust my skill in soldering and I don't know if that led is important (probably not, but I'm not sure).

romoloman, 2020-12-14:
The led should be desoldered, there is no way to disable it (have a lok to the schematics), but for running really low power there are other mods that should be made:
1) A pullup resistor is required between 3.3V and the CS pin on the flash (without that the flash will be always on draining 3 mA
2) The two LDO on board have a quiescent current of about 40uA each, replace them with RT9080 33GJA (2 uA each and the chips are pin to pin compatible)
3) The CP2104 is also draining 200 uA even if not used you need to cut the power supply line to the chip

After these mods I was able to achive less then 30uA in deep sleep mode.

if 318uA is sufficient for you just desolder the led and put a 100K resistor between pin 1 ad 8 of the flash.

I hope Xinyuan makes this mods on the board, maybe making a new version of the board

clonesht, 2021-01-04:
There is a non-hw way to reduce flash memory power usage:

> pinMode(13, OUTPUT);
> digitalWrite(13, HIGH);
> gpio_deep_sleep_hold_en();

[source: reddit](https://old.reddit.com/r/esp32/comments/idinjr/36ma_deep_sleep_in_an_eink_ttgo_t5_v23/ghx00sa/)

@romoloman do you have any stability issues with the RT9080-33GJ5 LDO mod?

According to a [comment](https://web.archive.org/web/20201112030913/https://blog.voneicken.com/2018/lp-wifi-esp32-boards/) it may drop out on peaks:

> Daniel deBeer
> [..]
> I tried RT9080 and it drops out during wakeup. Again the transient response from low to high current is not great.
> 
> There's a pitfall in LDO stability. Not all LDO's can work with large output capacitors and low ESR caps. Many LDOs have a minimum capacitor ESR or the part becomes unstable. I don't use those parts because at low temperature the capacitor ESR drops a lot and the minimum ESR is not a datasheet parameter for the cap, only the maximum, if its specified at all.

As alternative I have not tried, but [there is some positive experiences](https://www.esp32.com/viewtopic.php?t=6667) on [SOT23-5 version of Holtek HT7833](https://www.holtek.com.tw/documents/10179/82844a36-1633-498c-a6e3-4d99f2d0f6d7) and looks like it have the same pin assignment. (4mA LQC / 500mA out)

lewisxhe, 2021-01-21:
The blue one is the charging indicator, you can remove it without needing to.

closed 2021-01-21 by lewisxhe