Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #5: LiPo Voltage Measurement v2.3

Anton2k, 2021-01-04:
Hi, having a hard time trying to figure out how to measure the voltage of the LiPo battery, anyone had success?

Thanks,

Anton2k, 2021-01-04:
Update, I have now found that there is a Pin 35 BAT_TEST, that I can use to measure voltage, the issues is the reading is 2000+ when I was expecting it to me 3.9v I suspect the reading is in MV or something and I need to run it through some calculation?

RalfJL, 2021-01-05:
Have a look at the schematics.
There are 2 100kOhm resistors to prevent the input voltage from being higher than 3.3 V.
So you have to multiply the reading by 2 to get the correct measurement.

By the way, there are plenty of complains on the internet that the ADC is not very good on a ESP32.
Google it and you will find some code to smooth it a little bit. 
And don't take the measurement when BLE or WiFi is on. Readings become worse.

Anton2k, 2021-01-07:
Just encase anyone is looking, here us my solution...

`  esp_adc_cal_characteristics_t adc_chars;
  esp_adc_cal_value_t val_type = esp_adc_cal_characterize((adc_unit_t)ADC_UNIT_1, (adc_atten_t)ADC_ATTEN_DB_2_5, (adc_bits_width_t)ADC_WIDTH_BIT_12, 1100, &adc_chars);
  float measurement = (float) analogRead(35);
  float battery_voltage = (measurement / 4095.0) * 7.05;
`

closed 2021-01-07 by Anton2k

RalfJL, 2021-01-09:
I would recommend the following video
https://www.youtube.com/watch?v=UAJMLTzrM9Q
Normally I am happy with an average of 10 measurements and some correction I found on the internet
So my solution is more that:
double rawVoltage() {
  // read analog and make it more linear
  double reading = analogRead(VOLT_PIN);
  for (int i = 0; i < 10; i++) {
    reading = (analogRead(VOLT_PIN) + reading * 9) / 10;
  }
  if (reading < 1 || reading > 4095) return 0;
  return -0.000000000000016 * pow(reading, 4) + 0.000000000118171 * pow(reading, 3) - 0.000000301211691 * pow(reading, 2) + 0.001109019271794 * reading + 0.034143524634089;
}

All that without wifi or BLE