Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #6: T5 V2.2 2.9" schematics is wrong

RalfJL, 2021-01-05:
Hi,
the schematics for the above boards seams to be wrong.
The red LED is on with
digitalWrite(26, LOW);
So I assume it has the same connection as all the other boards:
+ -> Resistor -> LED -> PIN26

lewisxhe, 2021-01-21:
The green one is the power indicator, and the red one can be controlled by GPIO26. The green light can be removed if it is not needed.

closed 2021-01-21 by lewisxhe

L-oled, 2022-04-24:
Hi @RalfJL , I was facing to similar issue on the same board when trying to debug with openOCD (issue to find JTAG pinout):
it seem that there is "differents" v2.2 revision of this board.... My board have a JP1 20 pin connectors pinout that corresponds to the one here: http://www.lilygo.cn/prod_view.aspx?TypeId=50061&Id=1397&FId=t3:50061:3
But I didn't manage to find schematics, including on this directory (v2.2 don't match). Except for JP1 pinout, my board looks like a v1.2