Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #8: Under voltage & Overcharge Protection

Anton2k, 2021-01-08:
Hi, does the T5 v2.3 have built in cut off protection when using Lipo / Ion cells? and does it also have over charge protection? If so can you let me know what values these are, both cut off and over charge. I don't want to blow cells up or discharge them to the point they die.

Thanks,

Anton2k, 2021-01-10:
?

lewisxhe, 2021-01-21:
LILYGO T5 2.3 uses the TP4056 charging chip, which has charging protection, but it does not have reverse connection protection. When you use it, be careful not to mistake the power polarity.

closed 2021-01-21 by lewisxhe

Anton2k, 2021-01-21:
@lewisxhe thanks, but does it have cut off protecting, I need to know at what voltage does discharging stop at, 2.5v 3v?

Anton2k, 2021-01-29:
@lewisxhe ?

lewisxhe, 2021-01-29:
Battery over-discharge protection voltage: 2.5V  @Anton2k

haldi4803, 2021-02-08:
So you guys can confirm that battery works for you?
Because on my T5 v2.2 20181116 does not work when powering via Batteryconnector. I've plugged a battery, i've connected a 4v PSU but the T5 wouldn't even boot up :( No Green light, no Programm start. Do i have faulty hardware?