Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #13: T5 V2.2 2.9" consuming too much power

RalfJL, 2021-04-29:
Hi,
I removed the resistor in front of the green led,
 putting display driver to deep sleep,
putting esp32 to deep sleep
but still the board is consuming 9,35 mA.
This is unacceptable if you want to run it on a battery.
So what is left:
- the UART chip is there any way to turn it off
- the SD-card chip can I put that into deep sleep
- the battery charging chip is there any need to put it in power save mode

Is there any way to reduce the power consumption even more?

joysfera, 2021-05-16:
Seems like the CP2104 itself consumes at least 7 mA. I've read that adding a pulldown to its D- helps a lot.

joysfera, 2021-05-16:
It's possible to lower the consumption in sleep by about 2 mA by disabling flash:
```
pinMode(16, OUTPUT);
digitalWrite(16, HIGH);
gpio_deep_sleep_hold_en();
```
Still 8 mA left.

RalfJL, 2021-05-17:
Hi joysfera,
this is a great hack. Unfortunately everything is so tiny that I can't solder a pulldown resistor to anything.
So unless TTGO changes the layout, I can't see how this board can be powered from battery. 
This is a real shame because I put quite some effort to move away from TCP/IP and WiFi to ESP NOW which is much more power saving.
All in vain because of the board itself. 

So let's hope that TTGO is learning from this.

Thanks for sharing this information

Ralf