Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #15: Problem with pixels in last row of T5 v2.13

jcgfsantos, 2021-05-24:
I can't change the pixels below what looks to be line 122.
This is strange because the visible width for GxDEPG0150BN is 122.
Could this be a firmware or hardware problem?

Thanks

[image]

jcgfsantos, 2021-05-24:
I did some more tests and it looks like there is an offset.

I set display.setCursor(0, 0);
and then do:

    int i = 0;
    while (i < display.width()+1) {
        display.fillRect(0, 0, i,display.height(), GxEPD_BLACK);
        i++;
        display.update();
        delay(500);

        Serial.println(i);
    }

It starts drawing the rectangle immediately above those pixels that I can't change and reaches the top before the counter reaches display.width(). What can be causing this?

icepie, 2021-07-31:
I have same error too (T5 v2.13)

jcgfsantos, 2022-04-18:
@icepie, thanks!, your suggestion also worked for me, although the behaviour of the screen changes a little (it goes black sometimes).