Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #16: Intel CPU with MacOS 11.4 (Big Sur) can not upload anything to T5 board

solocatz, 2021-06-15:
For me only affect Lilygo product, my other low cost esp32 clones are fine.

Try to upload GetChipID sample:

esptool.py v3.0-dev
Serial port /dev/cu.usbmodem526F0051341
Connecting........__
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:47:3b:2c
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)

Here is a link of discussion. 
[ESPTool does not work on TTGO product](https://github.com/espressif/esptool/issues/280#issuecomment-861084553)

don41382, 2021-06-15:
I have the same issue with my Macbook M1 Air on Big Sur 11.4.

Please fix!

wecodemx, 2021-06-30:
Same issue here on Macbook intel on Big Sur 11.4  
Module: LILYGO® TTGO T-Call V1.4 / V1.5 ESP32

don41382, 2021-06-30:
Good news. After updating with new driver CP102 v6.0.1 from Silabs - it's finally working on my M1.

[Silabs Page](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers)

Let me know, if this works for you.

elajf, 2021-07-01:
> Good news. After updating with new driver CP102 v6.0.1 from Silabs - it's finally working on my M1.
> 
> [Silabs Page](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers)
> 
> Let me know, if this works for you.

This didn't work for me but I think mine has a different UART chip, a CH1902. Seems that could be the problem.

matteocarnelos, 2021-07-15:
Same problem here

losdevnull, 2021-07-19:
Same problem here.  Any workaround directly on Mac instead of trying to do this in Windows/linux VMs?

closed 2022-06-01 by lewisxhe