Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #20: lewisxhe / GxEPD boards.h is missing definition for T-Display E-Paper

ShotokuTech, 2021-08-29:
In  lewisxhe / GxEPD boards.h, line 213 change:
#elif defined(LILYGO_EPD_DISPLAY)
to:
#elif defined(LILYGO_EPD_DISPLAY_102)
And this resolves the problem.

LilyGO, 2021-09-11:
Please create a new board. h file, copy the following content into it

#pragma once

#if defined(LILYGO_T5_V213)

#define EPD_MOSI                (23)
#define EPD_MISO                (-1)
#define EPD_SCLK                (18)
#define EPD_CS                  (5)

#define EPD_BUSY                (4)
#define EPD_RSET                (16)
#define EPD_DC                  (17)

#define SDCARD_CS               (13)
#define SDCARD_MOSI             (15)
#define SDCARD_MISO             (2)
#define SDCARD_SCLK             (14)

#define BUTTON_1                (39)
#define BUTTONS                 {39}

#define BUTTON_COUNT            (1)

#define LED_PIN                 (19)
#define LED_ON                  (LOW)

#define ADC_PIN                 (35)

#define _HAS_ADC_DETECTED_
#define _HAS_LED_
#define _HAS_SDCARD_

#elif defined(LILYGO_T5_V22)

#define EPD_MOSI                (23)
#define EPD_MISO                (2)
#define EPD_SCLK                (18)
#define EPD_CS                  (5)

#define EPD_BUSY                (4)
#define EPD_RSET                (12)
#define EPD_DC                  (19)

#define SDCARD_CS               (13)

#define BUTTON_1                (37)
#define BUTTON_2                (38)
#define BUTTON_3                (39)

#define BUTTONS                 {37,38,39}
#define BUTTON_COUNT            (3)

#define LED_PIN                 (26)
#define LED_ON                  (HIGH)

#define ADC_PIN                 (35)

#define SPERKER_PIN             (25)

#define LEDC_CHANNEL_0          (0)
#define _HAS_ADC_DETECTED_
#define _HAS_LED_
#define _HAS_SPEAKER_
#define _BUILTIN_DAC_
#define _USE_SHARED_SPI_BUS_
#define _HAS_SDCARD_

#elif defined(LILYGO_T5_V24)
[... 173 lines cut ...]
#define EPD_RSET                (17)
#define EPD_DC                  (2)
#define EPD_BACKLIGHT_PIN       (0)

#define EPD_POWER_ENABLE        (5)

#define BUTTON_1                (35)
#define BUTTON_COUNT            (1)

#define ADC_PIN                 (34)
#define MOTOR_PIN               (4)

#define GPS_TX_PIN              (22)
#define GPS_RX_PIN              (21)
#define GPS_1PPS_PIN            (19)
#define GPS_RESET_PIN           (23)
#define GPS_WAKEUP_PIN          (18)

#define _HAS_ADC_DETECTED_
#define _HAS_POWER_CONTROL_
#define _HAS_GPS_

#else
#error "Please select model !!!"
#endif

carl3, 2022-02-09:
The examples inconsistently use LILYGO_EPD_DISPLAY and LILYGO_EPD_DISPLAY_102 . The "T-Display E-Paper 1.02inch" https://www.aliexpress.com/item/1005002536380666.html with a neopixel RGB LED uses LILYGO_EPD_DISPLAY.

Some of the examples use LILYGO_EPD_DISPLAY_102 instead of LILYGO_EPD_DISPLAY (e.g. hello_world) and some use both (incorrectly, as in GxEPD_Example.ino). In the lewisxhe repo the  Apr 14, 2021  update changed LILYGO_EPD_DISPLAY to LILYGO_EPD_DISPLAY_102 in some examples, but was not added to boards.h. You can define both LILYGO_EPD_DISPLAY_102 and LILYGO_EPD_DISPLAY, but some examples will cause problems.

Probably the boards.h should use ` #elif defined(LILYGO_EPD_DISPLAY) || defined(LILYGO_EPD_DISPLAY_102)` but most examples seem to use LILYGO_EPD_DISPLAY.

lewisxhe, 2022-08-08:
If there is still a problem, please reopen the issue

closed 2022-08-08 by lewisxhe