Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #23: The T-Display 1.02 E-Paper schematic is missing

carl3, 2022-02-01:
Can you upload the T-Display E-paper 1.02 schematic? (Aliexpress item 1005002536380666) It would be useful to examine power consumption, input voltage limits, and charging based on the board parts. (The product page links to this github repo.) Perhaps this is the EPD display, but the product on this is also not linked in the README.md

carl3, 2022-06-22:
It looks like the schematic added is the T92 with a 2.66 inch display, seems the same as the T5wth 2.66. The schematic for the T-Display E-paper 1.02 schematic is still missing and not linked. Thanks for updating the readme with a link to the T-Display product.

lewisxhe, 2022-08-08:
The schematic has been added, if there is still a problem, please reopen the issue

closed 2022-08-08 by lewisxhe