Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #25: Duplicate getUTF8Width definition in lewisxhe/GxEPD causes compile errors

carl3, 2022-02-09:
There are compile errors because getUTF8Width() has duplicate definitions in the lewisxhe/GxEPD fork mentioned as a prerequisite. There is a pull request from https://github.com/bot1131357/GxEPD that fixes the bug by commenting out the duplicate definitions.

The issue cannot be reported on the  lewisxhe/GxEPD fork because issues are disabled.

If you get this compile error, then you would need to use the https://github.com/bot1131357/GxEPD version.

reppr, 2022-02-21:
This issue caused me a lot of troubles...

Why are issues disabled in the lewisxhe fork?
I cannot understand why the problem was there since sept 2021 and is still not fixed.
What should people buying this board do with it, if the supplier cannot supply working software for it?
For me it was a frustrating experience...

@LilyGO 
Did LilyGo consider sending a free sample board to the main author of arduino ePaper software?

See also  https://forum.arduino.cc/t/compiling-error-on-lilygo-2-13-epaper-display-depg0213bn/955580/5

tkybundle, 2022-02-22:
I downloaded the lib here, so I was able to make 2.66 work.
https://github.com/Xinyuan-LilyGO/E-Paper-watch/tree/main/lib
anyway yeah, that sucks

lewisxhe, 2022-06-01:
I've been away for a while, since the repository is not maintained, the conflict has now been fixed, it's related to the update of the SDK

closed 2022-06-01 by lewisxhe