Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #28: Busy Timeout

Ahmedriazku2, 2022-07-12:
We have lillygo T5 board with 7.5 epaper. With a modification[Modified](https://drive.google.com/file/d/1f_AohgxSAYPVPwo2bbNI2PsKe3KxwP5s/view?usp=drivesdk)  of one spi pin 19/2 we successfully ran Waveshare[Original](https://drive.google.com/file/d/1fPD9XUesdmvl4mZLZVy-TptLoHgot-wX/view?usp=drivesdk)  weather program for two months. But since last ten days something happened, the screen went dimmed and on uploading new data sometimes it uploads but sometimes it gets stuck for days on Busy Timeout. I approached arduino forum where ZinggJM says he wont deal with Lillygo. Please do help.

lewisxhe, 2022-07-13:
You mean it's been working fine for two months? Sudden problems ten days ago?

Ahmedriazku2, 2022-07-13:
Now forget about GxEPD2 and busy timeout. Now when i run your own suggested example file the board is not updated. This is the file  i run: https://drive.google.com/open?id=1guJt0ITGjHKHFWHkqln3H-xrKVm_VhB-&authuser=ahmedriazkudept%40gmail.com&usp=drive_fs
its the same as your example https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/examples/GxEPD_Hello_world/hello_world.ino

now the output is this:
setup
setup done
Hello World

nothing is uploaded on the epaper. 

Now first fix this please. And please dont ask more questions.

lewisxhe, 2022-07-13:
I'm so sorry, this problem occurred

You can try to update the compiled binary executable to see if it is a hardware or software problem. Link:
https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/firmware/T5_V24_BW_GxDEPG0750BN.bin

Please use espressif flash_download_tools to flash.

https://www.espressif.com.cn/en/tools-type/flash-download-tools

For programming settings, please refer to the readme

https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/tree/master/firmware

lewisxhe, 2022-07-15:
If it still cannot be displayed after refreshing the program, it is very likely that the board is damaged. Please send the uploading program to the board, and then manually reset the board to AliExpress customer service. It is convenient for us to deal with you after sales.

lewisxhe, 2022-08-08:
If you have any other issues, please reopen the issue

closed 2022-08-08 by lewisxhe