Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #29: Problem achieving 300uA deepsleep current on T5 2.13" v2.3.1

mgsb, 2022-07-18:
I am trying to reproduce the stated 300uA deepsleep current for the 2.13" T5 version 2.3.1 with no LEDs and the DEPG0213BN display http://www.lilygo.cn/claprod_view.aspx?TypeId=62&Id=1391&FId=t28:62:28 . I am measuring 760uA using the deepsleep example from the ESP32-C3 T-OI-Plus repo https://github.com/Xinyuan-LilyGO/LilyGo-T-OI-PLUS/blob/main/example/deepsleep/deepsleep.ino . Do you have an example I can try to reproduce your results? ESP-IDF or Arduino/PlatformIO code?

lewisxhe, 2022-07-19:
The following two sleep modes occupy different sleep currents. I remember that the sleep current of ext1 is the lowest.
```
esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_TIMER);
esp_sleep_enable_ext1_wakeup(GPIO_NUM, ESP_EXT1_WAKEUP_ALL_LOW);
esp_sleep_enable_ext0_wakeup(GPIO_NUM, 0);
```

lewisxhe, 2022-08-08:
If you have any other issues, please reopen the issue

closed 2022-08-08 by lewisxhe