Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #30: T5 2.9” V2.2 Deep Sleep Issue

cvocvo, 2022-08-06:
I’m using ESPHome with a T5 2.9” v2.2 epaper, and when I enter deep sleep — shortly after entering deep sleep my screen becomes weird — half white / half black display with artifacts. I will try to post a picture.

The refresh will occur in deep sleep but the display itself won’t recover. 

Any ideas? Is deep sleep supported on v2.2?

cvocvo, 2022-08-06:
[image]

lewisxhe, 2022-08-08:
T5_V2.2 supports deep sleep. If the screen is refreshed after deep sleep is set, please check what is the reason for waking up. I don't use ESPHome. I will rule out the reason for the time being. you can refresh

https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/examples/GxEPD_Example/GxEPD_Example.ino

Check if you can sleep normally

cvocvo, 2022-08-08:
I think you're spot on -- I played with the refresh settings a little bit and found a post with a similar issue here: https://github.com/esphome/issues/issues/2334

Once I get it dialed in, I'll post some code this week or this weekend.

cvocvo, 2022-08-11:
I think I got it mostly sorted out; battery life still isn't great but that's a known limitation of v2.2 of the 2.9" board.
Here's what I've got for code -- I solved a bunch of problems (screen reloading issue, deep sleep, sleeping the flash, button to reload, etc.) and posted code up here: https://github.com/cvocvo/esphomeprojects/blob/main/epaper29.yaml

closed 2022-08-11 by cvocvo