Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #31: T5 V213 wich version ?

echelon75, 2022-09-11:
Hi ,
Here is a list of different hatdware/Dirver and I would like to know how to identify the right version on my hardware to flash the right Bin ?
T5_V213_BW_GxDEPG0213BN.bin
T5_V213_BW_GxGDEH0213B72.bin
T5_V213_BW_GxGDEH0213B73.bin

T5_V213_BW_GxGDEM0213B74.bin

T5_V213_BW_GxGDEW0213M21.bin

Regards

Richard

lewisxhe, 2022-09-11:
If purchased recently, try T5_V213_BW_GxDEPG0213BN.bin

echelon75, 2022-09-11:
Hi from France,

unfortunaly I have different versions since you this product has been launched.
I enjoy if you provide a small tutorial about *.bin and hardware version . 
Here is a picture of my board if it can help you : 
[](https://photos.app.goo.gl/uPqHNm16o16tKwTW6)

Regards 

Richard

lewisxhe, 2022-09-12:
If so, then you can try one by one, flushing the wrong bin file will not damage the hardware, it will just display abnormality, until the display is normal, then you may be that model