Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #32: Battery charge %

alfwro13, 2022-09-19:
Hi,
I have LILYGO® TTGO T5 V2.3 2.13 Inch E-Paper Screen and I have connected 3.7v 300mAh battery to the onboad battery connector.
I can see that on the board text written next to the battery connector BAT_TEST_35. What is it and can I use that to calculate battery % charge?

lewisxhe, 2022-09-21:
Yes, IO35 can collect battery voltage

alfwro13, 2022-09-26:
and is there a way to calculate the battery % charge ?

lewisxhe, 2022-09-28:
You can refer to this code

https://github.com/lewisxhe/XPowersLib/blob/master/src/XPowersAXP192.tpp#L1441-L1441

BatsIhor, 2022-12-07:
`   float voltage = analogRead(35) / 4096.0 * 7.46;
    uint8_t percentage = 100;
    if (voltage > 1)
    { // Only display if there is a valid reading
        Serial.println("Voltage = " + String(voltage));
        percentage = 2836.9625 * pow(voltage, 4) - 43987.4889 * pow(voltage, 3) + 255233.8134 * pow(voltage, 2) - 656689.7123 * voltage + 632041.7303;
        if (voltage >= 4.20)
            percentage = 100;
        if (voltage <= 3.50)
            percentage = 0;
        Serial.println("Percentage = " + String(percentage));
}`

BatsIhor, 2023-09-05:
@lewisxhe I checked your link but it's not clear how to use XPowerLib and what is the purpose of it. Do you have any documentation on what it is and when to use it? I was not able to find anything on the internet yet.

lewisxhe, 2023-09-06:
No, I'm just giving you a rough calculation of the percentage. Although it's not accurate, it can achieve the desired effect.