Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #36: Boards.h

Rolzzz, 2022-12-10:
All of your examples fail with the following

"Please edit Boards.h with a hardware abstraction for this board"

lewisxhe, 2022-12-10:
The sketch is compatible with multiple boards, each board has different pin definitions, and the model used needs to be defined above the sketch.
https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/4ad1172a7157b26640b1d359717f24ae34e61af7/examples/GxEPD_Example/GxEPD_Example.ino#L6

Rolzzz, 2022-12-10:
> The sketch is compatible with multiple boards, each board has different pin definitions, and the model used needs to be defined above the sketch.
> 
> https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/4ad1172a7157b26640b1d359717f24ae34e61af7/examples/GxEPD_Example/GxEPD_Example.ino#L6

I got GxEPD_Example.ino compiled and uploaded but my e-paper isn't functioning, I've emailed support to assist before I can try again

parahooners, 2022-12-28:
Hey - Try and roll back your board manager in Arduino - to 2.0.1 this worked for me

parahooners, 2022-12-28:
i also found that the below worked well with my screen even tho all the setups say use another 

#include <GxGDEH0213B73/GxGDEH0213B73.h>  // 2.13" b/w  old panel , form GoodDisplay

Rolzzz, 2022-12-28:
> I got GxEPD_Example.ino compiled and uploaded but my e-paper isn't functioning, I've emailed support to assist before I can try again

support got back to me and showed me to load right FW for board and I got screen working again.

For the Boards.h issue, I also got this worked out.

There are multiple B(b)oards.h files, one is upper case and one is lower case.

once I had both files in different areas read by Arduino IDE plus I had to comment out the first 2 lines of this set and uncomment the next lot then it all worked.

[image]

closed 2022-12-28 by Rolzzz