Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #40: Identify LilyGo-T5-Epaper version -> not available on website nor board

maxbrito500, 2023-01-26:
There exists no identification of the version either on the purchase page nor the board.

There is also no straightforward way to identify the screen model.

The problem is that guessing the board and screen is **not** working. Furthermore, this is a 4.7" screen and there is no driver mentioned there for that model. They are all smaller or bigger.

Purchase page is here: https://www.aliexpress.com/item/1005002272417292.html

If someone can help it would be appreciated.

Developers:
Please consider outputting the board version/details through the serial port with the default firmware. This would already save a lot of difficulty to get this information.

Many thanks!

lewisxhe, 2023-01-27:
The board you buy should correspond to this link:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47

closed 2023-01-27 by lewisxhe

maxbrito500, 2023-01-27:
Oh my. Thank you!

I got confused and assumed this was the repository for all ePaper modules. Topic solved.