Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #41: [Linux] Module needed to flash T5?

Shohreh, 2023-02-26:
Hello,

I ordered a ["DEPG0213BN 9102 Chip" T5](https://www.aliexpress.com/item/32869729970.html), altough it says "CH9102F" on the actual hardware I got, and it says "CH340" when connected to Linux.

I can reflash the board successfully on Windows using a web-based flasher.

On Linux, after successuflly compiling a project, it fails flashing the board: "Detecting chip type... Unsupported detection protocol, switching and trying again... Detecting chip type... ESP32-S2".

Is a module needed on Linux?

Thank you.

```
-bash-5.1# lsusb
Bus 001 Device 003: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 1a86:55d4 QinHeng Electronics USB Single Serial
Bus 002 Device 002: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

-bash-5.1# ll /dev/ttyU*
crw-rw---- 1 root dialout 188, 0 Feb 26 20:47 /dev/ttyUSB0

-bash-5.1# idf.py flash /dev/ttyUSB0
Executing action: flash
Serial port /dev/ttyUSB0
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running ninja in directory /root/esp/esp-idf/AirMeter.IO.Firmware/build
Executing "ninja flash"...
[1/7] Performing build step for 'bootloader'
[1/1] cd /root/esp/esp-idf/AirMeter.IO.Firmware/build/bootloader/esp-idf/esptool_py && /root/.espressif/python_env/idf5.0_py3.9_env/bin/python /root/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /root/esp/esp-idf/AirMeter.IO.Firmware/build/bootloader/bootloader.bin
Bootloader binary size 0x57a0 bytes. 0x1860 bytes (22%) free.
[2/5] cd /root/esp/esp-idf/AirMeter.IO.Firmware/build/esp-idf/esptool_py && /root/.espressif/python_env/idf5.0_py3.9_env/bin/python /root/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /root/esp/esp-idf/AirMeter.IO.Firmware/build/partition_table/partition-table.bin /root/esp/esp-idf/AirMeter.IO.Firmware/build/main.bin
main.bin binary size 0xde0f0 bytes. Smallest app partition is 0x100000 bytes. 0x21f10 bytes (13%) free.
[3/5] cd /root/esp/esp-idf/AirMeter.IO.Firmware/build/esp-idf/main && /root/.espressif/python_env/idf5.0_py3.9_env/bin/python /root/esp/esp-idf/components/spiffs/spiffsgen.py 0x1e000 /root/esp/esp-idf/AirMeter.IO.Firmware/device /root/esp/esp-idf/AirMeter.IO.Firmware/build/dev.bin --page-size=256 --obj-name-len=32 --meta-len=4 --use-magic --use-magic-len
[4/5] cd /root/esp/esp-idf/AirMeter.IO.Firmware/build/esp-idf/main && /root/.espressif/python_env/idf5.0_py3.9_env/bin/python /root/esp/esp-idf/components/spiffs/spiffsgen.py 0xc8000 /root/esp/esp-idf/AirMeter.IO.Firmware/webui/distgz /root/esp/esp-idf/AirMeter.IO.Firmware/build/web.bin --page-size=256 --obj-name-len=32 --meta-len=4 --use-magic --use-magic-len
[4/5] cd /root/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/root/esp/esp-idf -D "SERIAL_TOOL=/root/.espressif/python_env/idf5.0_py3.9_env/bin/python;;/root/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/root/esp/esp-idf/AirMeter.IO.Firmware/build -P /root/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x10000 main.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x22e000 web.bin 0x2f6000 dev.bin
esptool.py v4.5
Serial port /dev/ttyUSB0
Connecting..........

A fatal error occurred: This chip is ESP32-S2 not ESP32. Wrong --chip argument?
CMake Error at run_serial_tool.cmake:55 (message):

  /root/.espressif/python_env/idf5.0_py3.9_env/bin/python;;/root/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32
  failed

FAILED: CMakeFiles/flash
cd /root/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/root/esp/esp-idf -D "SERIAL_TOOL=/root/.espressif/python_env/idf5.0_py3.9_env/bin/python;;/root/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/root/esp/esp-idf/AirMeter.IO.Firmware/build -P /root/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /root/esp/esp-idf/AirMeter.IO.Firmware/build/log/idf_py_stderr_output_10554 and /root/esp/esp-idf/AirMeter.IO.Firmware/build/log/idf_py_stdout_output_10554
```

lewisxhe, 2023-02-27:
On my Ubuntu, it is not ttyUSB
, but ttyACM0, please check whether the correct port is selected
[image]

Shohreh, 2023-02-27:
Thanks for the tip. dmesg -wH does show a /dev/ttyACM0 device popping up.

But flashing fails, regardless of the board being ON or OFF:

```
echo "alias get_idf='. $HOME/esp/esp-idf/export.sh'" >> ~/.bashrc
source ~/.bashrc

get_idf

cd ~/esp/esp-idf/AirMeter.IO.Firmware
-bash-5.1# idf.py /dev/ttyACM0 flash
Executing action: /dev/ttyACM0
ninja: error: unknown target '/dev/ttyACM0'
command "/dev/ttyACM0" is not known to idf.py and is not a Ninja target
```

lewisxhe, 2023-02-27:
Are you using idf? You should specify the port like this

```
idf.py -p /dev/tty.ttyACM0 -b 115200 flash

```

Shohreh, 2023-02-27:
Flashing does seem to work with the command I used.

lewisxhe, 2023-02-27:
If the problem is solved, the problem will be closed

closed 2023-02-27 by lewisxhe