Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #44: Unable to communicate with T5 v2.3.2 using T-U2T

StevenCellist, 2023-08-08:
Hi guys, I got my hand on a brand-new TTGO T5 v2.3.2 with 2.13 inch e-ink display (not yet released in lilygo.cc but available through reseller).
As per my reseller, the difference w.r.t. v2.3.1 is the need to use the T-U2T to program the board.
This is confirmed by plugging a USB-A to USB-C cable straight into the board: Windows Device Manager doesn't show any port, while the board is being powered and running the example sketch.

When I plug just the T-U2T into my USB cable, Device Manager shows `USB-SERIAL CH340K (COM10)`, indicating that the adapter is working nicely. However, when I plug it into the board, I get no communication at all. When I reset the board using RST button, nothing. When uploading a sketch and pressing BOOT on the U2T, "no serial data received". When holding BOOT and pressing and releasing RST, "no serial data received". I tried another machine, another cable (which did work on other USB-C boards) - no bit of luck.

How should I go about communicating with and programming this device? What could I try? Is there any documentation for this new version 2.3.2?

Attached are pictures of the board, in case that someone wants to look at it.
[image]
[image]

StevenCellist, 2023-08-09:
@lewisxhe thanks for your reply.  I uninstalled the drivers for the specific COM port through Device manager, rebooted and installed your linked drivers. I should note that the installer also shows "CH34SER.INF" as driver name.
Once I connect the U2T, the CH340K name comes back up. And per [this](https://forum.arduino.cc/t/lilygo-ttgo-ch9102f-chip-windows-drivers/949744/5) post, that should work just fine. But still no response from the T5 v2.3.2 :(
Could there be anything else that I'm doing wrong?

lewisxhe, 2023-08-09:
Do you use Windows?

lewisxhe, 2023-08-10:
try this driver
[CH9102.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/files/12307856/CH9102.zip)

StevenCellist, 2023-08-10:
Did the following:
- uninstall device and check "remove drivers" from Device Manager
- remove U2T
- reboot
- install the .exe
- reboot
- plug in U2T
- Device Manager at first doesn't recognize, just lists "USB Serial"
- after a couple seconds, it apparently installs the drivers and again: CH340K
:( sorry to make this so complicated.. PC is stubborn!

lewisxhe, 2023-08-10:
If it recognizes 340K, then it is really possible that it is 340K.

You try to insert U2T into the board and connect it to the computer via USBC, then open Serial, and click the RST of the board to see what information is output.

lewisxhe, 2023-08-10:
Can the USB screen be updated by inserting it?

StevenCellist, 2023-08-10:
What exactly do you mean by that?
The device does power up and shows the out-of-the-box example animation of squares in the corners, showing all available characters etc.

lewisxhe, 2023-08-10:
Okay, contact the seller and have them send you a new U2T.

StevenCellist, 2023-08-12:
Yes, the 'old' one was not correct! I got sent a new one - the sticker on the box is marked completely different and it shows up as CH9102, and I can successfully read serial information now!

In case anyone ends up here: my incorrectly working one had the box marked with H494, while the working one is marked with Q320 and CH9102F as well.

@lewisxhe thanks for the debugging and thinking along!!

closed 2023-08-12 by StevenCellist