Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #46: SD card  not working when using T5_V2.3.1_213]

bruno1950, 2023-09-26:
Hi,
I just bought  a Lilygo T5 V2.3.1.213 version GDEM0213B74 CH9102 and I was able to show on display data from I2C sensors (BME280 and SHT3x) and to get a timestamp from RTC DS3231. Deepsleep also works fine. However, I was not able to use the SD card, neither inserting it directly in the slot nor connected to an external SDcard reader ( MOSI IO15,MISO IO2, SCK IOIO14, CS IO13). The card is correctly mounted  but I keep on gettin an error "opening the file" or ""appending the file". Even using a vey simple sketch just to try if it's possible to write something on the SD card is unsuccesful. Since I think that Lilygo is a good brand I have to assume that it's my fault but I was not able to find my mistake. Could anyone tell me which code I have to use just to write data on SD card in order to use it as a datalogger ? Thx in advance

lewisxhe, 2023-09-27:
Please copy your sketch here

bruno1950, 2023-10-01:
Hi,

sorry for the long delay in answering your request but I was in hospital for a programmed surgery that luckily went fine.

Here you have the sketch that doesn't work for what attains the SD card since on the serial monitor I get the message "opening file failed" and "appending file failed" and no data are saved on the SD

When I use this sketch ( clearly with needed modifications) wth the TFT Lilygo board it works just fine.

Thx for your hekp 

Bruno

#include <SD.h>
#include <FS.h>
 
#include <SPI.h>
#include <Wire.h>
 

#include <RTClib.h> // for the RTC
RTC_DS3231 rtc;
 
#include <GxFont_GFX.h>
 
#include <GxEPD2.h>
#include <GxGDEM0213B74/GxGDEM0213B74.h> // specific gor the Lilygo eink display model I have
 
#include <Fonts/FreeMonoBold9pt7b.h>
 
#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>
 
#define SPI_MOSI 23
#define SPI_MISO -1
#define SPI_CLK 18
#define ELINK_SS 5
#define ELINK_BUSY 4
#define ELINK_RESET 16
#define ELINK_DC 17
 
#define SDCARD_SS 13
#define SDCARD_CLK 14
#define SDCARD_MOSI 15
#define SDCARD_MISO 2
 
GxIO_Class io(SPI, /*CS=5*/ ELINK_SS, /*DC=*/ ELINK_DC, /*RST=*/ ELINK_RESET);
GxEPD_Class display(io, /*RST=*/ ELINK_RESET, /*BUSY=*/ ELINK_BUSY);
 
SPIClass sdSPI(VSPI);
 
bool sdOK = false;
 
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
 
#define SEALEVELPRESSURE_HPA (1013.25)
Adafruit_BME280 bme; // I2C
 
#include <SHT3x.h>
SHT3x Sensor;
 
// variables to store data from sensors ( float type because they are value with a decimal part)
 
float TEMP;
float HUM;
float temp;
float hum;
 
String timeStamp;
 
[... 195 lines cut ...]
delay(1500);

display.update();
 
Read_TempHum();
displayReadings();
logSDCard();
 
// increasing the counter for the next data readings
 
sensor_data++;

Serial.println("Sensor data logged successfully! Going to sleep");
 
// ESP32 going to sleep for the amount of time you decided
esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
delay(5000);
esp_deep_sleep_start();
 
delay(2000);
esp_deep_sleep_start();
}
 
void loop()
{}

lewisxhe, 2023-10-04:
It seems that there is no problem with the SD part. Has the SD card been formatted in FAT32 format? Is the capacity less than 32G? Have you replaced any other SD tests?

bruno1950, 2023-10-04:
Hi, thx for your prompt feedback.
I tried with two different SDcard-adapters and with two different SD cards, one with 4 Gb and the other with 16 Gb, both formatted FAT32, but I still have the same error "Failed to open file for appending" 
If I use codes devised to test SDcard (such as the SD_test.ino) I get the error message "Card Mount Failed" both when I insert the card in the slot of the board and if I use an external SD-card reader connected according to instructions.
At this point  one question and one request. 
Question: Have you tried to run my sketch ( or a similar one with data collected from an I2C sensor) with SD card on a Lilygo T5_V2.3.1_213 ? Were you successful ?
Request: Would you be so kind to send me  a working code for using SD card  as a data logger with this board ? I was not able to find any simple example of this use  and I need to save on SDcard data collected from an I2C sensor.
Otherwise this board is of no use for me.
 Thx for your help.

lewisxhe, 2023-10-07:
Use the factory test file to test to see if it is normal @bruno1950 
[firmware](https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/tree/master/firmware)

bruno1950, 2023-10-17:
Hi,
I tried to use the file factory test as you suggested but I was not able to get a result even following your example.
For firmware I downloaded the [T5_V24_BW_GxGDEM0213B74.bin] because my board is this type. Then I downloaded from Espressif site the ESP32 FLASH DOWNLOAD TOOL but the available version is different from your example and I am blocked. I'm enclosing the pictutes of what I'm getting.
I've also tried to use your SD_test.ino but I got a lot of errors and it doesn't work.
Since I'm a newbie could you you please send me a code for SD card working for my Lilygo board ? or at least telling me where I can find it ?
I've seen that my issue was raised also last year by another user and seems NOT to have been solved (https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/issues/33)

Thanks in advance for your help. The board would be perfect IF the SD card worked
[image]
[image]

lewisxhe, 2023-10-19:
You have burned the wrong firmware, you should have burned 
[T5_V213_BW_GxGDEM0213B74.bin](https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/firmware/T5_V213_BW_GxGDEM0213B74.bin)