Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #47: I cannot clear T5 2.31_2.13 display

zsiroslaci, 2023-12-13:
I made a program which wrote text on it. But the text written on display cannot be removed. New text written by new program over write it so the old text pixels and the one  both displayed. The new text  is removable, but the old one cannot. 

What can I do to remove old text?   

Br
Laszlo

lewisxhe, 2023-12-15:
Is it as if the font is permanently printed on the screen?

zsiroslaci, 2023-12-15:
Yes. It is permanently printed on the screen.Whath can I do with it?BrLaszlo Küldte a Windows Posta Feladó: Lewis HeElküldve: 2023. december 15., péntek 3:10Címzett: Xinyuan-LilyGO/LilyGo-T5-Epaper-SeriesMásolatot kap: zsiroslaci; AuthorTárgy: Re: [Xinyuan-LilyGO/LilyGo-T5-Epaper-Series] I cannot clear T5 2.31_2.13 display (Issue #47) Is it as if the font is permanently printed on the screen?