Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #48: [DEPG0266BN] spiram: SPI RAM not initialized

iw0der, 2023-12-13:
Hi,
I'm experiencing some errors on the hardware module LILYGO® TTGO T5 2.66-Inch E-Paper. Now I want to report here the first problem encountered!

I try to use/enable the external RAM via Arduino IDE but I only get crashes, at most I get to this:

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
E (298) spiram: SPI RAM not initialized

abort() was called at PC 0x401027b7 on core 0

Backtrace: 0x400838cd:0x3ffe3af0 0x4008d1cd:0x3ffe3b10 0x400926cd:0x3ffe3b30 0x401027b7:0x3ffe3bb0 0x401027f7:0x3ffe3bd0 0x401026cb:0x3ffe3bf0 0x400e5623:0x3ffe3c20 0x40082fe1:0x3ffe3c50 0x40079316:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: ee06d735f4dcf31a
```

I chose the ESP32 Wrover Kit as the compilation template but I also tried other combinations... I chose psram enable but when I run the code it gives an error like the one reported.

Am I forgetting some steps or do I have a problem with recognizing the external RAM?

At boot it seems that it is not detected and the crash reports library functions even before my setup:
```
0x400838cd: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x4008d1cd: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x400926cd: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c line 46
0x401027b7: esp_spiram_get_chip_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c line 145
0x401027f7: esp_spiram_get_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c line 241
0x401026cb: esp_himem_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/esp_himem.c line 134
0x400e5623: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c line 201
0x40082fe1: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/cpu_ll.h line 39

```

Thanks

73,
Arturo.

iw0der, 2023-12-13:
With board "ESP32 Wrover kit (all version)" and an if that uses psramInit() it doesn't crash, but it still says it can't see the external RAM!

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[   451][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
NET: Connecting to WiFi ....
NET: Connected to AP successfully!
NET: IP=10.11.12.13, RSSI=-72
EPD-Test_2c13b83
UTIL: ESP32 Partition table:

| Type | Sub |  Offset  |   Size   |       Label      |
| ---- | --- | -------- | -------- | ---------------- |
|  01  | 02  | 0x009000 | 0x005000 | nvs              |
|  01  | 00  | 0x00E000 | 0x002000 | otadata          |
|  00  | 10  | 0x010000 | 0x140000 | app0             |
|  00  | 11  | 0x150000 | 0x140000 | app1             |
|  01  | 82  | 0x290000 | 0x160000 | spiffs           |
|  01  | 03  | 0x3F0000 | 0x010000 | coredump         |
| ---- | --- | -------- | -------- | ---------------- |

[  2236][I][util.cpp:186] partitionTable(): max APP size: 1280 kB
[  2238][I][util.cpp:202] psRAM(): ========================================
[  2245][W][util.cpp:217] psRAM(): PSRAM is not enabled on this board
[  2251][I][util.cpp:219] psRAM(): ========================================
[  2257][I][util.cpp:220] psRAM(): Internal RAM  size   : 321880
[  2263][I][util.cpp:221] psRAM(): Internal RAM  free   : 251136
[  2269][I][util.cpp:222] psRAM(): ========================================
[  2276][I][util.cpp:258] initID(): ESP32: chip=1, cores=2, revision=3, radio=WiFi/BT/BLE flash=4MB external

```

lewisxhe, 2023-12-15:
It seems that the PSRAM has failed. Contact the seller to replace it.