Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #49: Sleep current on T5 2.13" v2.3.1

ldab, 2024-01-05:
Related to https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/issues/29

When using the example [examples/GxEPD_Example/GxEPD_Example.ino](https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/examples/GxEPD_Example/GxEPD_Example.ino) the sleep current is 635uA from when measuring from the VBAT pin.

[image]

How was the 300uA measured?

lewisxhe, 2024-04-10:
Measured on the onboard CP2104 UART
[image]

claudobahn, 2024-11-14:
@lewisxhe I'm having the same issue as @ldab with the same board (T5 2.13" v2.3.1 w/ DEPG0213BN), the lowest I've been able to achieve w/ sleep current is around 600-700 micro amps.

Is your test running with an unmodified board? What sketch are you using? And are you isolating or resetting any of the GPIO before going to deep sleep? Changing any pull-up/down settings?

I'm wondering if there's current drain because of a pull-up/down resistor, either internal to the ESP32 or on the T5 board.

lewisxhe, 2024-11-15:
I did not make any changes, and the example was also tested using https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/examples/GxEPD_Example/GxEPD_Example.ino. The only difference is the current test tool.