Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #52: No binary for v2.66

digidolphin, 2024-04-08:
I think I might have bricked my device by using a wrong display definition in code i compiled. I need to upload the original binary to reset the device (i think), but cannot find a binary for v2.66 - only t92 and v2.4 2.66" - and they do not seem to work. Please provide the correct binary.

lewisxhe, 2024-04-10:
T92 IS https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/blob/master/firmware/T92_V1.0_BW_GxDEPG0266BN.bin

digidolphin, 2024-04-11:
Thank you. Unfortunately, I have tried that and it does not work. My device still does not change anything on the screen when plugging in USB. Any suggestions on how to fix this?

[image]

lewisxhe, 2024-04-11:
Yeah ? Then try this and see if it's normal
[H493.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series/files/14942986/H493.zip)

digidolphin, 2024-04-16:
Any suggestions on how to make it work again?