Xinyuan-LilyGO/LilyGo-T5-Epaper-Series #61: T5 2.13 burned out?

HenryData, 2025-01-01:
Hi, I was trying to upload the sketch from https://github.com/jasonful/Tides/tree/master when at some point the screen greyed out and the T5 stopped communicating. 
I've been trying to flash the binary file back to it but it just goes to sync for awhile before I click stop and it outputs 
no log file output ...
I have an adafruit TPL5110 board connected connected to pin 13, gnd, and Vbat. And a LI battery connected to the ground and VDD pins on the TPL5110. The TPL5110 still lights up when I plug in the USB to the board or connect the battery to the TPL5110. 
I cant reset the screen or upload any code. I guess I must have burnt out the hardware but I'm not sure how.
Is it possible that having the USB and the battery connected at the same time damaged something?
Thanks
[image]

lewisxhe, 2025-01-02:
Connecting the battery and USB at the same time will not damage the board.

Are you unable to upload now?

Find IO0, connect it to GND, then power on, and finally write the firmware to try

HenryData, 2025-01-03:
After cleaning off a corner of flux and doing this it seems to be unstuck. Still having trouble uploading the sketch but thats a seperate issue and at least the screens responding. Thanks!

closed 2025-01-03 by HenryData